opendroneid/opendroneid-core-c #3: Integrate European Harmonization Changes to Spec

gabrielcox, 2019-06-01:
Implement the following changes:

1. Change of N/S, E/W speed components to Speed/(compass) Direction values.  The encoding is totally re-done for this.
2. Change Height above takeoff to Height (with option to set "above takeoff" or "AGL")
3. Add support for multiple "Self-ID" messages each with a separate type. Currently, there are 2: 
    0: Description
    1. Remote Pilot ID

friissoren, 2019-06-27:
Item 3 has been done by PR #2. Item 1 and 2 are still pending.
For item 1, the N/S Direction bit has not yet been changed in the specification to be the E/W direction bit. Implement this as E/W since the specification should be changed.

friissoren, 2019-07-29:
Item 1 and 2 are covered by PR #7

closed 2019-08-01 by friissoren