opendroneid/opendroneid-core-c #9: About implementing opendroneid

stevenroelofsen, 2019-08-05:
Hi, 

INVOLI would like to implement the opendroneid standard in its system. We were wondering if it is possible to have access to the apps to validate our implementation of the standard.

Best regards.

friissoren, 2019-08-06:
Thanks for contacting. Great to hear that someone else is also interested in this :-)

I am not completely sure what you refer to with the term "apps"? Could you possibly clarify?

So far, the only opensource component related to OpenDroneID (that I am aware of) is this Github project, which provides just the core encoding and decoding functions.

There is ongoing work to expand the Mavlink message definitions to support suitable messages for feeding data to/from the opendroneid-core-c functions: https://github.com/mavlink/mavlink/pull/1167

The OpenDroneID standard itself is not completely frozen yet.

stevenroelofsen, 2019-08-06:
> functions: mavlink/mavlink#1167

Hi,

Thank you for answering. I tried to contact you at admin@opendroneid.org
but the email could not be delivered.

In the video on https://www.opendroneid.org/ there are two mobile devices
(an iPhone and an Android) with some app. We were wondering if we could
have access to those apps for validation.

We plan to do our prototype on a Raspberry Pi 3B+ with something like a
Alfa AWUS036NHA for WiFi and a nRF52840 Dongle for Bluetooth. I am not sure
if Mavlink would be useful, but if there is something on how to forge
OpenDroneID packets on Linux that would be great.

As for the standard itself, we are aware that it is not finished yet, but
we are at the prototype stage our-self so it isn't a problem.

Best regards,

Steven

friissoren, 2019-08-06:
I see. Those two apps were developed earlier for internal purposes at Intel. I need to check whether we are allowed to distribute those in binary form or not. I can't promise anything at this point.

That being said, there definitely should be arranged some kind of interoperability testing between different implementations. I have made an inquiry whether this is something that is already being planned or not.

friissoren, 2019-08-13:
I discussed with Gabriel Cox about this. We will start the internal process of trying to make the Android and iOS apps open source. This is unfortunately a rather bureaucratic process, so let's see how long that will take. 

However, this might not actually help you very much. I understood that you are working on a WiFi implementation. The apps currently only support Bluetooth receiving. Gabriel will put you in contact with a person that is working on a WiFi implementation.

Gabriel will fix the email address on opendroneid.org.

friissoren, 2020-01-23:
Sorry for the very long delay. As expected, getting through the bureaucracy in Intel took quite a while but the source codes for the Android app have now been on Github for a few weeks:
https://github.com/opendroneid/receiver-android
Only Bluetooth receiving is supported though. There are persons who have stated interest in implementing support also for WiFi receive. Let's see what happens with that.

closed 2020-01-23 by friissoren