opendroneid/opendroneid-core-c #35: wifi.h

sxjack, 2020-09-13:
Please consider renaming wifi.h odid_wifi.h. I am trying to get this going on an ESP32 and wifi.h causes a conflict.

friissoren, 2020-09-14:
@gabrielcox @simonwunderlich @nicolas-s : What do you think?

This would cause some interruption for anyone using this code, since they would have to change the name of the include file. On the other hand, it is a fairly small change and the ESP32 is a quite likely chip for people to use when implementing drone ID transmitters.

I assume you are doing a WiFi implementation? If you are only interested in Bluetooth, there is a CMake flag (BUILD_WIFI) that will leave out the WiFi part.

BTW: @sxjack: If you get the WiFi/Bluetooth transmit part running on an ESP32 and you are willing to open source that implementation, that would be a very valuable contribution to this overall open drone ID project. We could create a new repository here for that.

sxjack, 2020-09-14:
The WiFi side looks straightforward, my main problem being that the function to transmit a raw packet doesn't seem to be available under Arduino so I will have to move to the full SDK. 

I am trying to get my head around the bluetooth side of things.

I will probably open source the code. I am currently waiting to see what the UK government do regarding electronic ID/conspicuity of recreational UAVs.

simonwunderlich, 2020-09-14:
The wifi.h is currently only used internally in the library, so there shouldn't be much reason to rename it. On the other hand, we can export it and then it definitely makes sense to rename it to not clash. So I'm in favor of renaming it now, as long as it's only used internally not too many people should care.

@sxjack would you be willing to prepare a pull request for that?

BTW, the sender example is made to send out the action frame through send_nl80211_action(), which is nl80211 and only supported in Linux, but you can use odid_wifi_build_message_pack_nan_action_frame() to create the action frame and send it through a different API. I don't know which APIs the ESP32 supports for sending raw frames, though.

sxjack, 2020-09-14:
The ESP32 has esp_wifi_80211_tx() for sending raw frames, but it doesn't appear to be visible using Arduino.

I am a git novice, but I'll have a look at doing the pull.

sxjack, 2020-09-15:
Pull done (I think).

I have also modified sender/main.c to set the valid flags and get it to compile without errors on a Raspberry Pi.

nicolas-s, 2020-09-15:
I agree with simon. It's an internal header and should not be a problem when compiling opendroneid-core-c as a static library.
I don't know the SDK and the toolchain of ESP32 but if I look at the documentation they talk about the possibility to compile "[components](https://docs.espressif.com/projects/esp-idf/en/release-v3.1/api-guides/build-system.html#concepts)" (modular pieces of standalone code which are compiled into static libraries).

Also the SDK headers for the wifi part seem to be called esp_*.h. https://github.com/espressif/esp-idf/tree/master/components/esp_wifi/include
The only one called wifi.h is in fact always referenced as "esp_private/wifi.h" https://github.com/espressif/esp-idf/tree/master/components/esp_wifi/include/esp_private.

I think it's possible to compile opendroneid-core-c as is, but maybe I don't see the whole problem. 
If you really think it's impossible I have no problem renaming it.

friissoren, 2020-09-15:
Thanks. I merged PR #35.

closed 2020-09-15 by friissoren