opendroneid/opendroneid-core-c #38: Build fails on Ubuntu 20.04

janusw, 2021-01-24:
Trying to build on Ubuntu 20.04 (Focal Fossa) ...

```
$ cmake .
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libgps'
--   Found libgps, version 3.20
-- Checking for module 'libnl-genl-3.0'
--   Found libnl-genl-3.0, version 3.4.0
-- Configuring done
-- Generating done
```

... fails with the following error:

```
[ 46%] Building C object wifi/sender/CMakeFiles/sender.dir/main.c.o
/home/janus/github/opendroneid/opendroneid-core-c/wifi/sender/main.c: In function ‘drone_adopt_gps_data’:
/home/janus/github/opendroneid/opendroneid-core-c/wifi/sender/main.c:232:36: error: invalid operands to binary * (have ‘timespec_t’ {aka ‘struct timespec’} and ‘int’)
  232 |  time_in_tenth = gpsdata->fix.time * 10;
      |                  ~~~~~~~~~~~~~~~~~ ^
      |                              |
      |                              timespec_t {aka struct timespec}
```

I assume this is due to the fact that 20.04 has the updated libgps version 3.20. On Ubuntu 18.04, with libgps version 3.17, everything works well.

janusw, 2021-01-24:
The [release notes](https://gpsd.gitlab.io/gpsd/NEWS) of gpsd/libgps mention for release 3.20:
```
Convert all timestamp_t to timespec_t.
```
The change has been implemented in this [commit](https://gitlab.com/gpsd/gpsd/-/commit/1d4a09541a18c453c5586e6f57d3eb5fec9526ed).

So it seems like 3.20 is the breaking release, and I assume that 3.18/3.19 should still work.

janusw, 2021-01-24:
Made an attempt to fix this in 24248efef49aa465f8a3390f7070537f4dfd5a28. Feedback welcome.

friissoren, 2021-01-25:
Thanks for spotting and fixing this.

closed 2021-01-25 by friissoren

janusw, 2021-01-25:
Thanks for merging :)

PR #39 (2021-01-25): fix build errors with libgps 3.20 and newer
* fixes #38
* happens e.g. on Ubuntu 20.04
* support for older libgps versions is maintained
* depending on the libgps version, we define a preprocessor flag
  and use conditional compilation to distinguish both cases
* the TSTONS macro is taken from:
  https://gitlab.com/gpsd/gpsd/-/blob/1d4a09541a18c453c5586e6f57d3eb5fec9526ed/timespec.h#L102