opendroneid/opendroneid-core-c #40: Compiler warnings with gcc 8.4

janusw, 2021-01-25:
I'm building on Ubuntu 20.04 with gcc 8.4:

```
$ cmake . -DCMAKE_C_COMPILER=gcc-8
-- The C compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/gcc-8
-- Check for working C compiler: /usr/bin/gcc-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libgps'
--   Found libgps, version 3.20
-- Checking for module 'libnl-genl-3.0'
--   Found libnl-genl-3.0, version 3.4.0
-- Configuring done
-- Generating done
```

The build runs through fine, but there are several compiler warnings, all of the same type:

```
[ 53%] Building C object test/CMakeFiles/odidtest.dir/opendroneid_sim.c.o
/home/janus/github/opendroneid/opendroneid-core-c/test/opendroneid_sim.c: In function ‘ODID_getSimData’:
/home/janus/github/opendroneid/opendroneid-core-c/test/opendroneid_sim.c:98:51: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
             strncpy(basicID_data.UASID, id, sizeof(id));
                                                   ^
/home/janus/github/opendroneid/opendroneid-core-c/test/opendroneid_sim.c:134:53: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
             strncpy(auth_data.AuthData, data, sizeof(data));
                                                     ^
/home/janus/github/opendroneid/opendroneid-core-c/test/opendroneid_sim.c:143:58: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
             strncpy(selfID_data.Desc, description, sizeof(description));
                                                          ^
/home/janus/github/opendroneid/opendroneid-core-c/test/opendroneid_sim.c:166:67: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
             strncpy(operatorID_data.OperatorId, operatorId, sizeof(operatorId));
                                                                   ^
[ 61%] Building C object test/CMakeFiles/odidtest.dir/test_inout.c.o
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c: In function ‘test_InOut’:
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c:54:38: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(BasicID.UASID, id, sizeof(id));
                                      ^
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c:85:47: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(Auth0.AuthData, auth0_data, sizeof(auth0_data));
                                               ^
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c:93:47: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(Auth1.AuthData, auth1_data, sizeof(auth1_data));
                                               ^
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c:100:45: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(SelfID.Desc, description, sizeof(description));
                                             ^
/home/janus/github/opendroneid/opendroneid-core-c/test/test_inout.c:121:54: warning: argument to ‘sizeof’ in ‘strncpy’ call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
     strncpy(operatorID.OperatorId, operatorId, sizeof(operatorId));
                                                      ^
```

They do not occur with gcc 7.5 or earlier. It should be checked if they are justified (or just false alarm). If not, one could just disable the warning flag.

friissoren, 2021-01-26:
I managed to install Ubuntu 20.04 on Windows Subsystem for Linux (that was a learning experience in itself) to try to reproduce this. For some reason that comes with gcc version 9.3.0, which only warns about some of the warnings you list above. I am not sure why. Anyway, those are easily fixable and are in the test code, so not that critical.
In that environment there were a lot of warnings from the Mavlink code, and since I don't want to start changing anything in there, I added a compile flag to suppress those.
At the same time, I noticed multiple lint warnings shown in the CLion editor, so I fixed those as well. 

PR #41 opened. Please review
@gabrielcox, @lukasbrchl

janusw, 2021-01-26:
> I managed to install Ubuntu 20.04 on Windows Subsystem for Linux (that was a learning experience in itself) to try to reproduce this.

Yeah, WSL is pretty cool. Finally one gets a proper operating system inside this funny thing called "Windows" ;)

> For some reason that comes with gcc version 9.3.0, which only warns about some of the warnings you list above. I am not sure why.

Yes, 20.04 comes with gcc 9 by default, but you can install other versions side-by-side. I also tried gcc 9, but for me it produced even more warnings (on top of those reported above), so I thought I'd start with those reported by gcc 8 :)

janusw, 2021-01-26:
> PR #41 opened. Please review

Thanks for taking care of this! I have not actually looked at your changes, but I can confirm that the branch fixes all warnings for gcc-8 and gcc-9.

However, I actually do still get a linker error with gcc-10 [`gcc version 10.2.0 (Ubuntu 10.2.0-5ubuntu1~20.04)`], as available in Ubuntu 20.04:
```
[ 84%] Linking C executable odidtest
/usr/bin/ld: CMakeFiles/odidtest.dir/test_inout.c.o:(.bss+0x370): multiple definition of `operatorID_enc'; CMakeFiles/odidtest.dir/opendroneid_sim.c.o:(.bss+0xa0): first defined here
```

It can be fixed by using the flag `-fcommon`, as documented here: https://gcc.gnu.org/gcc-10/porting_to.html

friissoren, 2021-01-27:
Thanks. I see what the problem is and I made a quick fix. If you could maybe verify that this now links without any errors?

janusw, 2021-01-27:
> Thanks. I see what the problem is and I made a quick fix. If you could maybe verify that this now links without any errors?

Yes, I can confirm that 560716cdee475f05413d2c2a8f17d9e7e60fd57f fixes the linker error with gcc 10. Thanks!

closed 2021-02-01 by friissoren