opendroneid/opendroneid-core-c #50: Development environment

sxjack, 2021-11-05:
What are you using to develop this? 

libopendroneid used to compile on my Raspberry Pi, but I am getting errors with the latest code.

error: dereferencing pointer to incomplete type ‘struct ieee80211_vendor_specific’
cc1: warning: unrecognized command line option ‘-Wno-address-of-packed-member’

gcc 8.3.0

friissoren, 2021-11-05:
I just checked that it compiles fine with the following versions of gcc/g++: 7.5.0, 8.4.0 and 9.3.0

I am compiling on Ubuntu 20.04 currently.

Could you show the full compile error message?

sxjack, 2021-11-05:
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c: In function ‘odid_wifi_build_nan_sync_beacon_frame’:
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c:303:22: error: dereferencing pointer to incomplete type ‘struct ieee80211_vendor_specific’
     if (len + sizeof(*vendor) > buf_size)
                      ^~~~~~~
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c: In function ‘odid_wifi_build_message_pack_beacon_frame’:
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c:471:22: error: dereferencing pointer to incomplete type ‘struct ieee80211_ssid’
     if (len + sizeof(*ssid_s) > buf_size)
                      ^~~~~~~
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c:483:22: error: dereferencing pointer to incomplete type ‘struct ieee80211_supported_rates’
     if (len + sizeof(*rates) > buf_size)
                      ^~~~~~
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c:493:22: error: dereferencing pointer to incomplete type ‘struct ieee80211_vendor_specific’
     if (len + sizeof(*vendor) > buf_size)
                      ^~~~~~~
/home/sjack/src/opendroneid-core-c/libopendroneid/wifi.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
make[2]: *** [libopendroneid/CMakeFiles/opendroneid.dir/build.make:76: libopendroneid/CMakeFiles/opendroneid.dir/wifi.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:91: libopendroneid/CMakeFiles/opendroneid.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

friissoren, 2021-11-05:
I was able to compile the ESP32 project after copying over the latest core-c code to the Arduino library folder (including wifi.c) and updating the copy of wifi.c with the clock changes and adding some missing basic ID enum values in the switch in the init function of id_open.cpp (which it for some reason complained about).

I am not sure what exact compiler the Arduino 1.8.13 editor environment is using.

Are you sure you have copied over all of the latest files from core-c?

sxjack, 2021-11-05:
Forget this. If looks like a weird filesystem problem on the Pi.

sxjack, 2021-11-05:
I don't get any errors (other than the usual one from the BLE library) with Arduino 1.8.13, ESP 32 1.0.4 (Windows 10).

sxjack, 2021-11-05:
> It looks like a weird filesystem problem on the Pi.

Actually it seems to have been me being an idiot.

friissoren, 2021-11-05:
No problem :-) Let's close this issue.

closed 2021-11-05 by friissoren