opendroneid/opendroneid-core-c #56: DroneCAN implementation

tridge, 2022-08-06:
I've implemented DroneCAN (previously known as UAVCAN) messages for OpenDroneID that mirror the MAVLink messages.
The message definitions are here:
https://github.com/dronecan/DSDL/tree/master/dronecan/remoteid
they deliberately use identical field names and field definitions to the MAVLink XML to make a dual-protocol implementation easy
I've done a dual-protocol implementation for Arduino on a ESP32-S3 here:
https://github.com/ArduPilot/ArduRemoteID
this implements MAVLink over UART and DroneCAN on bxCAN (known as TWAI in ESP32) in parallel, allowing the OpenDroneID squiter to be connected to a flight controller via either protocol. I've tested this implementation with ArduPilot and the OpenDroneID android receiver.
I think mentioning the DroneCAN definitions would be worthwhile in the OpenDroneID docs to avoid someone else having to re-invent them.

friissoren, 2022-08-09:
Absolutely. And thanks for the info. I updated the main README.

Just a side-note: You probably are aware of this already but the ESP32-S3 ArduRemoteID implementation will not be fully rule compliant in the US and the EU until the underlying SW gets updated to support transmitting also Bluetooth 5 Long Range signals (unless you rely on the Wi-Fi transmission methods, of course).

closed 2022-08-09 by friissoren

tridge, 2022-08-09:
yes, I hope to have BT5 next week

commit 896f9dbc (2022-08-09): Update readme with FAA approval, Ardupilot and DroneCan news

Fixes #56