opendroneid/opendroneid-core-c #57: Verification of Japanese rule content

friissoren, 2022-08-12:
I found a public version of what I believe to be the [Japanese rule for remote ID](https://www.mlit.go.jp/koku/content/001444589.pdf).

I found a way to auto-translate the document to English but the translation was far from clear in all places. I tried to generate a [summary](https://github.com/opendroneid/opendroneid-core-c#japan) of the main changes compared to the US and EU rules/standards and I also updated the comparison [table](https://github.com/opendroneid/opendroneid-core-c#comparison).

@ToshihiroMakuuchi: Would it be possible for you to read through the Japanese version of the rule document and correct any mistakes I have made?

ToshihiroMakuuchi, 2022-08-12:
@friissoren 
I am not very good at English, but I will help you as much as I can.
Could you give me a few minutes of your time?

ToshihiroMakuuchi, 2022-08-13:
@friissoren 

Once I checked, I wasn't sure if I could get the exact answer I intended regarding the data format.
We believe that this Issue should be formally answered by a government agency.
Therefore, we sent an email to the Ministry of Land, Infrastructure, Transport and Tourism, asking them to respond to this issue.

If possible, could you please send the same information for this question as well?

--------
Aviation Safety and Security Department, Civil Aviation Bureau,
Ministry of Land, Infrastructure, Transport and Tourism, Japan

hqt-jcab.remoteid@mlit.go.jp
--------

Best Regards,

friissoren, 2022-08-15:
Thank you. I essentially only have two main questions. I will post them here and also send an email to mlit.go.jp.

1) The Japan rule mandates that Bluetooth 5 Long Range is used (or Wi-Fi Beacon). However, it is unclear to me whether it is also mandatory to transmit Bluetooth 4 Legacy Advertising signals simultaneously with the BT5 Long Range signals, or whether the BT4 part is optional (or even not allowed at all)?

2) Since ASTM F3411-19 is the main standard reference, the System message is optional. This means that it is optional to transmit the Operator dynamic location/altitude. This surprised me, because both the US and the EU rule are mandating that this information is transmitted. Probably the Japanese government have not seen any need for transmitting the Operator location/altitude but I just wanted to double check that this is correct.

friissoren, 2022-08-26:
The Japanese government replied to my email query and confirmed that BT4 and operator location/altitude are optional.

ToshihiroMakuuchi, 2022-08-27:
I, too, have been contacted by the Japanese government. The following wording appears to be the correct one. Please reflect this.
----------
　　　　　　　　　　　　　　   Before　　         After
  Operator dynamic position      　O(?)　     ⇒　  　O
  Operator altitude WGS-84       　O(?)　     ⇒    　  O
  BT4 Legacy Advertising         M(?)⑤⑥(?)    ⇒　　  O
  BT5 Long Range                    M⑤⑥(?)       ⇒   　   M⑤

friissoren, 2022-08-27:
Great. Thanks. Those changes are done to the table.

commit 4dc54c6c (2022-08-26): Update Japan rule information

The Japanese government replied on a query to:
hqt-jcab.remoteid@mlit.go.jp

They confirmed that BT4 and the operator location/altitude
are optional.

Fixes #57