opendroneid/opendroneid-core-c #66: Difference between different message types

michaelzing, 2023-04-09:
Hi,

I was reading libopendroneid/opendroneid.h and libopendroneid/opendroneid.c and am a little bit confused by how it works. Could you explain why there are so many different types of encodings?

int encodeBasicIDMessage(ODID_BasicID_encoded *outEncoded, ODID_BasicID_data *inData);
int encodeLocationMessage(ODID_Location_encoded *outEncoded, ODID_Location_data *inData);
int encodeAuthMessage(ODID_Auth_encoded *outEncoded, ODID_Auth_data *inData);
int encodeSelfIDMessage(ODID_SelfID_encoded *outEncoded, ODID_SelfID_data *inData);
int encodeSystemMessage(ODID_System_encoded *outEncoded, ODID_System_data *inData);
int encodeOperatorIDMessage(ODID_OperatorID_encoded *outEncoded, ODID_OperatorID_data *inData);
int encodeMessagePack(ODID_MessagePack_encoded *outEncoded, ODID_MessagePack_data *inData);

From my understanding, to decode a message, you would want to first decode the MessagePack. But what is the difference between all the other encodings? Is it that all of the other messages must be contained in the MessagePack?

In addition, I believe the MessagePack must be broadcast for extended advertising? But for legacy advertising, do you just broadcast the individual messages one at a sequentially?

Thanks in advance, this is a great project 🙏

gabrielcox, 2023-04-09:
inData can contain a subset of the messageTypes in the standard, but it must contain "all message types being sent" (F3411-22a, 5.4.5.22).  For example: an FAA compliant implementation may only send the 3 required messages types of BasicID, LocationVector, and System.  Just send them all in a single messagePack each time rather that encoding 1 messagePack with BasicID and another messagePack with the rest.

Note that inData is of type ODID_MessagePack_data, and within ODID_MessagePack_data you'll see that it contains an array of messages.  Also be sure to be familiar with how union typedefs work (it's relevant in how the array of messages work within ODID_MessagePack_data.

For legacy broadcasts, you don't use messagePacks at all.  Just send the individual messages of each desired type in sequence.

michaelzing, 2023-04-10:
Thank you so much for your help! That clears it all up.

closed 2023-04-10 by michaelzing