opendroneid/opendroneid-core-c #69: Active Scanning for Wi-Fi Beacons

michaelzing, 2023-04-27:
Hi, I have a question that's loosely related to opendroneid, but I can't think of a better place to ask.

I'm developing a scanner for remote ID Wi-Fi beacons. Since the ASTM standard technically allows beacons to be broadcast on most channels in the 2.4 and 5.0 GHz bands, I want to be able to scan for beacons on all of these channels, therefore, I want to use Wi-Fi active scanning. (active scanning is where the scanner sends a probe request on a certain channel, and any devices that regularly broadcast on that channel will instantly send a probe response back with the beacon data. This way, the scanner doesn't have to passively wait for beacons to be broadcast, which could take up to the beacon interval amount of time). 

I'm wondering if drones or remote ID broadcast modules support active scanning? In other words, will they necessarily respond to a probe request that a scanner sends out?

friissoren, 2023-04-28:
I think that is dependent on the individual transmitter implementation. I do think some transmitters support this and thus appear to have a better receive rate in the Android receiver application but this will only work when the receiver is reasonably close to the transmitter (since the whole thing is then suddenly two-way communication instead of the normal one-way broadcast and most smartphones probably don't transmit with very high power levels).

michaelzing, 2023-04-28:
Thank you so much! That is really helpful. Is there any chance you know whether DJI drones have active-scanning enabled?

And do you have an estimate of how close "reasonably" close is (for the active scanning to work)?

michaelzing, 2023-04-28:
Also, if you don't mind me asking, do you know if Remote ID Wi-Fi beacons are allowed to be transmitted on channels in the 5 GHz band that are indoor only or DFS/TPC (dynamic frequency selection or transmit power control) required? 

The ASTM standard just says "The broadcast message shall (BWFB0030) be transmitted on either ... Or Any Wi-Fi channel of the 5150 MHz to 5895 MHz band".

gabrielcox, 2023-04-29:
> The ASTM standard just says "The broadcast message shall (BWFB0030) be transmitted on either ... Or Any Wi-Fi channel of the 5150 MHz to 5895 MHz band".

The F3411-22a standard is a general technical specification that must be adapted locally.  Different countries have differing regulations (especially in the 5Ghz band) that must be overlaid on the technical specification.  In the US, F3586-22 (The Means of Compliance (MOC) to FAA Part 89) must be adopted for the US Airspace.

A couple of clauses serve as this overlay:
1. Within F3411-22a, section 11.1: "...Otherwise, the stated requirements must be met 95 % of
the time unless other local regulations apply."

2. FAR Part 89.310(g)(1 & 2): The unmanned aircraft must be capable of broadcasting the
message elements in § 89.305 using a non-proprietary broadcast
specification and using radio frequency spectrum compatible with
personal wireless devices in accordance with part 15 of title 47,
Code of Federal Regulations, where operations may occur without
an FCC individual license.

So, the F3411 gives a technical standard under which RID may be performed, however, in any given locality, the implementation may be further constrained by communications and aviation regulations.  It's the combination of these overlays that you should expect of a Remote ID implementations.  Therefore, it would practical/optimal for a receiver implementation to exclude unnecessarily scanning indoor frequencies (where they are designated as such).

gabrielcox, 2023-04-29:
> I'm wondering if drones or remote ID broadcast modules support active scanning? In other words, will they necessarily respond to a probe request that a scanner sends out?

Adding to Soren's correct response:  
The design of RID is a one-way broadcast communication that is not limited by the transmit power/design of the device on the ground.  To meet the performance requirements of RID, the antenna design on the aircraft could be far superior to what may be integrated into a cell phone that is designed for different use cases (often close proximity devices).  

Therefore, active scanning wasn't built in as a requirement.  That said, some implementations may include it anyway, and, as long as the gratuitous broadcast requirements are being met (every second regardless of the presence of active scanners), those implementations will hopefully provide the best of both designs.

michaelzing, 2023-05-01:
Thank you so much Gabriel, this is incredibly helpful. Do you know enough about Wi-Fi DFS and TPC to also comment on how that applies to Remote ID? Since Remote ID only requires periodic broadcasts of beacons instead of continuously transmitting data like in a Wi-Fi network, would drones on channels that require DFS and TPC have to implement DFS and TPC?

In addition, would you happen to know if DJI drones have active scanning built in and will respond to probe requests?

closed 2023-09-13 by friissoren