opendroneid/opendroneid-core-c #74: Interest in support French Remote ID

twpayne, 2023-08-23:
France has its own WiFi Beacon-based Remote ID format described in https://www.legifrance.gouv.fr/jorf/id/JORFTEXT000039685188/.

Would you be interested in a PR that adds support for this format?

I would suggest adding a function similar to the existing `odid_wifi_build_message_pack_beacon_frame` function, maybe called `odid_wifi_build_french_beacon_frame`.

friissoren, 2023-08-24:
Certainly, if you are able to provide those.

Unfortunately, I personally cannot really allocate any significant time on this activity (or most other remote ID related improvements) anymore but my belief is that anything that can be open-sourced and shared with everyone will help lessen the burden of supporting the regulatory/mandatory aspects of remote ID.

BTW: Do you know the legal status of French remote ID? My earlier assumption has been that it would be replaced with the European version of remote ID (from ASD-STAN) to align with the general EU rules, but it is of course possible that France decides to stay with their current remote ID rules/laws?

twpayne, 2023-08-24:
Thanks for the info. I'll open a PR when I have some working code.

I've not been able to find out the legal status of the French remote ID. I also would be curious to know, especially how it interacts with EU-wide legislation.

PR #76 (2023-09-13): Add support for French Remote ID
Fixes #74.

This PR adds initial support for building French Remote ID beacon frames.

The French system encodes a slightly different set of parameters in a different way, so the shared code with Open Drone ID is mainly around the construction of the beacon frame packet. I've re-used the Open Drone ID way of representing missing values.

The implementation has been verified with Wireshark and a French Drone ID dissector.
closed 2023-09-13 by friissoren