opendroneid/opendroneid-core-c #84: Is the ESP32-S3-MINI1 legal to be used in the real world ?

DavideLentini, 2023-12-29:
Hi all , I own an esp32-s3-mini1 and found the opendroneid firmware a great alternative to use remoteid .
I am in Europe , and from Jan 1 2024 , the drone I own will have to have a remote id module .
So I would like to know if legally I can use the esp32-s3-mini1 as a remote id module , or do I have to buy a certified module .
I think if it was illegal , this repository would not even exist :) , so I think I already know my answer.
I hope someone knowledgeable can enlighten me . thanks

friissoren, 2024-01-04:
Sorry about the slow response.

Using the [ArduRemoteID](https://github.com/ArduPilot/ArduRemoteID) code for the ESP32-S3 will get you pretty far, but might not alone fulfill all of the requirements specified in the [EU rules and standards](https://github.com/opendroneid/opendroneid-core-c?tab=readme-ov-file#european-union).

E.g., if you are using an S3 module that has an onboard PCB antenna, it is unlikely to satisfy the requirements related to the omnidirectional transmission of the remote ID signals. There are possibly other cases like this. You would need to do your own homework and analysis of the rules and requirements.

Also, be aware that there are differences in the rules for modules that are fully integrated with the drone (i.e. getting location etc. information via MavLINK or similar from the flight controller) and fully stand-alone modules (with onboard GPS receiver, battery etc.).

closed 2024-01-04 by friissoren

gabrielcox, 2024-01-06:
I would like to add a thought. 

The use of this code is not an issue of legal/illegal. 

Legalities come in when either manufacturing a (compliant or non-compliant) drone product or operating a (compliant or non-compliant) drone product. They can also come in with FCC, CE, UL requirements subject to your geographical regulator.

These repositories contain source code tools that can be used to make a compliant product.  Use of these source tools *alone* will certainly not guarantee full compliance because there are many other compliance factors (beyond just the source code, like signal strength, channel usage, etc) that go into a final compliant product.  

There's ASTM standards in the U.S. and ASD-Stan standards (which include ASTM standards) in the E.U. That give you instructions on how to determine compliance of your product.

DavideLentini, 2024-01-07:
Thank you all for the response , you have been very kind. I will study the various papers to better understand how I should act in my situation . :)