opendroneid/opendroneid-core-c #89: Stack Canary Watchpoint Triggered (BTC_TASK) when using opendroneid-core-c library

butterisgod, 2024-03-14:
Description:
I am encountering a "Stack canary watchpoint triggered (BTC_TASK)" error when using the opendroneid-core-c library in my ESP32 project. The error occurs while attempting to decode and print Open Drone ID data from a received Bluetooth advertisement packet.

Steps to Reproduce:

Set up an ESP32 project with the Arduino IDE.
Include the opendroneid-core-c library in the project.
Implement the MyAdvertisedDeviceCallbacks class with the onResult callback function.
In the onResult function, check if the received advertisement packet is a valid drone beacon.
If it is a valid drone beacon, decode the Open Drone ID data using the decodeOpenDroneID function from the library.
Attempt to print the decoded Open Drone ID data using the printBasicID_data and printLocation_data functions.
Expected Behavior:
The code should successfully decode the Open Drone ID data from the received Bluetooth advertisement packet and print the relevant information without triggering any errors or exceptions.

Actual Behavior:
When running the code, a "Stack canary watchpoint triggered (BTC_TASK)" error occurs, causing a panic in the BTC_TASK. The error seems to happen while printing the Open Drone ID data, specifically after printing the UAType.

Error Message

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 

```
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK) 

Core  0 register dump:

PC      : 0x4038245f  PS      : 0x00060236  A0      : 0x80381b27  A1      : 0x3fca83f0  

A2      : 0x3fc98790  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060223  

A6      : 0x00060223  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  

A10     : 0x00000000  A11     : 0x3fcf0862  A12     : 0x0000000a  A13     : 0x3fc96fe0  

A14     : 0x02c98790  A15     : 0x00ffffff  SAR     : 0x00000011  EXCCAUSE: 0x00000001  

EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000
```
Environment:
ESP32-S3-Dev
Arduino 2.3.2
opendroneid-core-c library

Code used to produce this error 

```
#if not defined(ARDUINO_ARCH_ESP32)
#error "This program requires an ESP32"
#endif

#pragma GCC diagnostic warning "-Wunused-variable"
#pragma GCC diagnostic ignored "-Wunused-but-set-variable"

#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <BLEBeacon.h>
#include <WiFi.h>
#include "opendroneid.h"

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.printf("Bluetooth Device: %s\n", advertisedDevice.toString().c_str());
    
    // Print raw advertisement packet
    uint8_t* payloadPtr = advertisedDevice.getPayload();
    size_t payloadLength = advertisedDevice.getPayloadLength();
    
    Serial.print("Raw Advertisement Packet: ");
    for (size_t i = 0; i < payloadLength; i++) {
      Serial.printf("%02X ", payloadPtr[i]);
    }
    Serial.println();
    
    // Check if the advertisement is a valid drone beacon
    if (payloadLength >= 2 && payloadPtr[0] == 0xFA && payloadPtr[1] == 0x0B) {
      // Decode Open Drone ID data
      ODID_UAS_Data uasData;
      memset(&uasData, 0, sizeof(uasData));
      ODID_messagetype_t messageType = decodeOpenDroneID(&uasData, payloadPtr);
      
      if (messageType != ODID_MESSAGETYPE_INVALID) {
        Serial.println("Open Drone ID data:");
        if (uasData.BasicIDValid[0]) {
          printBasicID_data(&uasData.BasicID[0]);
        }
        if (uasData.LocationValid) {
          printLocation_data(&uasData.Location);
        }
        // Print other decoded data as needed
      }
    } else {
      Serial.println("Not a valid drone beacon");
    }
    
    Serial.println();
  }
};

BLEScan* pBLEScan;

void setup() {
  Serial.begin(115200);
  Serial.println("Open Drone ID Scanner");
  
  // Initialize Bluetooth
  Serial.println("Initializing Bluetooth...");
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);
  
  // Initialize Wi-Fi
  Serial.println("Initializing Wi-Fi...");
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
}

void loop() {
  // Perform Bluetooth scan
  Serial.println("Scanning for Bluetooth devices...");
  BLEScanResults foundBLEDevices = pBLEScan->start(5, false);
  Serial.print("Bluetooth devices found: ");
  Serial.println(foundBLEDevices.getCount());
  pBLEScan->clearResults();
  
  // Perform Wi-Fi scan
  Serial.println("Scanning for Wi-Fi networks...");
  int numNetworks = WiFi.scanNetworks();
  Serial.print("Wi-Fi networks found: ");
  Serial.println(numNetworks);
  
  for (int i = 0; i < numNetworks; i++) {
    Serial.printf("SSID: %s, RSSI: %d\n", WiFi.SSID(i).c_str(), WiFi.RSSI(i));
    
    // Print raw advertisement data for each Wi-Fi network
    Serial.print("Raw Advertisement Data: ");
    uint8_t* bssid = WiFi.BSSID(i);
    for (int j = 0; j < 6; j++) {
      Serial.printf("%02X ", bssid[j]);
    }
    Serial.println();
    
    Serial.println();
  }
  
  WiFi.scanDelete();
  
  Serial.println("Scan done!");
  delay(30000);
}
```

gabrielcox, 2024-03-15:
Could you please isolate this further to a single call (either by step debugging, or commenting/uncommenting/printing to figure out where it's failing).  This code is generic and intended for any platform, but may need some porting to any particular platform. 
 It is tested on Linux and Windows.  Since this is a source code forum, the request should come down to a source code change request / PR.  Have you researched the error message?  I just did and it seems that it could be running out of memory or running past a data buffer.  Have you tried limiting to only minimal libraries (like BT only) to see if that makes a difference?

gabrielcox, 2024-07-18:
@butterisgod Any updates on this?  Please see previous comment.

gabrielcox, 2026-02-11:
Closed with no continuing activity after 2 requests.