opendroneid/opendroneid-core-c #92: Python CFFI

afrs-us, 2024-09-03:
Attempting to utilize this C library in Python CFFI is an uphill battle w/ respect to slurping in the header due to complex math included in the header.

Thought about checking code out the source and moving the math into the functions VS allowing it to reside in the header - But before doing that, I wanted to propose a question to the team if anyone has had success using libopendroneid.so w/ Python on *nix OS?    If success on Python, did you use ctypes, cffi, or some other module?

In summary:
The math inside the opendroneid.h header file causes Python CFFI to reject the header.
Instructions within the readme for different platforms would be helpful.

I appreciate everyone who contributes to this project.

Thanks

friissoren, 2024-09-04:
Personally I have not made any such attempts. 

Once you boil it down, there isn't that much code in this repo and none of it requires any major processing power, so maybe it might even be easier to pick the parts that you need and create a completely new pure Python implementation?

Of course, there is then always the risk of introducing new errors/mis-translations.

If you do end up on that path, and wish to share the results, a new repo can easily be created in the opendroneid project for storing that.

gabrielcox, 2024-09-04:
Bluemark created an MQTT subscriber, and in the process built up a python implementation that may help with something like this: https://github.com/BluemarkInnovations/RemoteID-MQTT-subscriber .